ClockSelect/myevic #342: Wismec RX 2/3 keeps resetting to vw (power) mode

itssamb, 2017-12-02:
I have been using my wismec rx 2/3 for about 7 months now with Uwell crown tank without any issues whatsoever. I use my crown tank at around 240 C and 50 watts with .15 ohm coils and have been fairly satisfied with the flavor and vape.

However, I recently bought the new Uwell valyrian and have been unable to use this new tank in temp control mode. I have tried different wattage as well as temp configurations to no avail. As soon as I hit fire, the mod resets from tc to vw (power) mode. I've even switched back to the old crown tank to see if the mod had gone kaput but it works fine with that. Have even tried upgrading firmware, tried preheat but nothing has worked so far. Also, I am not too technical and barely manage with the basic settings of the mod so not sure on how to fix this. Please help!

MicroSur, 2017-12-02:
Look for coils material in valyrian, its Kanthal, not temp controlled wire.

itssamb, 2017-12-02:
That’s a bummer 😔. Even in vw mode, the Vape is too hot if I try it in the recommended wattage, ie 95 Watts, what can I do to get a warm but not too hot Vape

enricorov, 2017-12-02:
I second @MicroSur. You need specific materials (Ni200, Ti, SS) for TC. If you try to use any other wire in TC, the mod will sense something's up and revert to VW.